zenitheesc/Drivers #2: STM32: INA3221

MarcusMoraisEpifane, 2021-04-17:


leocelente, 2021-09-09:
Fiz algumas mudanças, mais só pra conseguir compilar. E chamei a `alive` dentro da `init`.

leocelente, 2021-09-28:
Deu pra ter uns avanços desde os testes de domingo. 
 - Os offsets BIT_xxxx de 3bits tavam contando da esquerda (MSB), mas é ao contrario
 - A conversão dos valores não tava levando em consideração o right shift e a extensão de sinal
 - O calculo da potencia tava usando só os valores do CH1

Mas acho que o chip é defeituoso mesmo. Escrever (qualquer valor) pro Configuration Register muda ele para 0x0027. Então os bits de enable dos canais são desativados. O jeito é não escrever configuração por enquanto. Quando estivermos com um chip original vamos testar de novo.

leocelente, 2021-09-28:
Como tá "funcionando" com a configuração padrão vou fechar esse issue, mas ainda é uma boa fazer aquelas modificações que a gente tinha falado domingo.

closed 2021-09-28 by leocelente