zenitheesc/Drivers #16: STM32: AS5047

MarcusMoraisEpifane, 2021-04-18:


leocelente, 2021-07-16:
## Funcional mas pode melhorar
No momento o código que existe detecta transições do sinal I usando polling. Em teoria, dá pra fazer o calculo da frequência com dois timers.

### Possível Implementação
Um timer periódico executando uma interrupt a cada _delta_tempo_ millisegundos. Tudo que essa interrupts faz é pegar o valor de um segundo timer.

Esse segundo timer é configurado como um contador, que acumula com pulsos externos
#### Pseudo código
```asm
counter_interrupt:
    nop                        ; implicito counter++

timer_done: 
    read counter               ; lê quantidade de voltas
    div  counter, delta_tempo  ; calcula velocidade
    write counter 0            ; reseta
```
#### Referencia
https://letanphuc.net/2015/06/stm32f0-timer-tutorial-and-counter-tutorial/

### Validação 
Continuar com o esquema de gerar a onda quadrada pelo Proteus

leocelente, 2022-02-01:
Não temos previsão de usar em projetos atualmente, vamos reabrir o issue quando for necessário

closed 2022-02-01 by leocelente