zenitheesc/Drivers #8: STM32: MAX31865

MarcusMoraisEpifane, 2021-04-18:


leocelente, 2022-02-01:
# Aplicação:  

O responsável por esse issue deve: 
 - **Pesquisar sobre essa aplicação**
    - Em qual contexto do projeto será utilizada? 
    - Qual geralmente é o propósito dessa aplicação? 
    - Existe algum dispositivo comum?
 - **Pesquisar sobre o dispositivo atual**
    - Por que estamos usando ele?
    - Tem algo que torna ele difícil de trabalhar?
    - Onde ele já foi utilizado?
 - **Implementar uma biblioteca do dispositivo.**
    - Qual informação está associada ao dispositivo?
    - Como se espera que ele seja utilizado?
    - Sua implementação compila sem Warnings?
 - **Documentar a Implementação**
    - O que cada função faz?
    - O que cada struct e seus campos significam?
    - Quais características do dispositivo afetam o código?
    - Quais partes do datasheet um usuário do driver deve ficar atento?
    - Existe alguma outra documentação externa de interesse para o usuário?

## Observações

## Tarefas
- [ ] Pesquisa e Leitura do Datasheet
- [ ] Selecionar Features
- [ ] Rascunho da Interface
- [ ] Implementação
- [ ] Documentação
- [ ] Revisão 
- [ ] Teste
- [ ] Finalização

leocelente, 2022-02-01:
### Situação atual
Estamos utilizando o CI: **MAX6682** para termistores. 
Como já existe uma biblioteca para as equações não lineares, o que falta ao Driver é a comunicação SPI