zenitheesc/Drivers #33: Driver para Barômetro

leocelente, 2022-01-25:
# Aplicação: Barômetro

O responsável por esse issue deve: 
 - **Pesquisar sobre essa aplicação**
    - Em qual contexto do projeto será utilizada? 
    - Qual geralmente é o propósito dessa aplicação? 
    - Existe algum dispositivo comum?
 - **Pesquisar sobre o dispositivo atual**
    - Por que estamos usando ele?
    - Tem algo que torna ele difícil de trabalhar?
    - Onde ele já foi utilizado?
 - **Implementar uma biblioteca do dispositivo.**
    - Qual informação está associada ao dispositivo?
    - Como se espera que ele seja utilizado?
    - Sua implementação compila sem Warnings?
 - **Documentar a Implementação**
    - O que cada função faz?
    - O que cada struct e seus campos significam?
    - Quais características do dispositivo afetam o código?
    - Quais partes do datasheet um usuário do driver deve ficar atento?
    - Existe alguma outra documentação externa de interesse para o usuário?

## Observações

## Tarefas
- [ ] Pesquisa e Leitura do Datasheet
- [ ] Selecionar Features
- [ ] Rascunho da Interface
- [ ] Implementação
- [ ] Documentação
- [ ] Revisão 
- [ ] Teste
- [ ] Finalização

GustavoKanaiama, 2022-01-25:
Jóia

leocelente, 2022-01-25:
:gem:

leocelente, 2022-01-25:
#31 foi um teste

PR #34 (2022-04-05): Driver do Barômetro (MS5607)

Closes #33 

## What type of PR is this? (check all applicable)
- [x] ✨ Feature
- [ ] 🐞 Bug Correction
- [ ] 📝 Documentation Update
- [ ] 🚩 Other

# Changes:

# Testing
closed 2022-04-05 by leocelente